ARM-software/lisa #1749: Issue: trace.ana.tasks.plot_task_residency() fails to plot for all domains

Leo-Yan, 2021-09-27:
**Describe the bug**
The system have two CPU freq domains, the first domain is CPU0/1/2/3, the second domain is CPU4/5/67/; When use notebook to analyze task behaviour, the method trace.ana.tasks.plot_task_residency() only can plot the task residency on the first CPU domain (CPU0/1/2/3), but it doesn't show up the task residency on the second CPU domain.

**To Reproduce**
Steps to reproduce the behavior:
* what part of LISA is being used (synthetic tests, WA-related features, etc.): synthetic tests
* what kind of target is being used (linux, android, etc.): Hikey960 with Android*
* what code is being run if that is a custom script using LISA APIs:
Please see the ipython script: https://gist.github.com/Leo-Yan/c1e33b0c399762dcbbe943fc632fbad5; you could see at the box 'In [84]:', it calls method trace.ana.tasks.plot_task_residency("tsk4_1-13"), the picture only show CPUs 0~3 but have no CPUs 4~7.

douglas-raillard-arm, 2021-09-27:
Could you run this and share the output please ?
```
print(target.plat_info)
```

As side notes I would recommend:
* using bokeh backend rather than matplotlib (unless you specifically need matplotlib for some reason). This will give you significantly better plots for interactive use:
```
import holoviews as hv
# Enable bokeh as default backend, for interactive plots
hv.extension('bokeh')
```
https://lisa-linux-integrated-system-analysis.readthedocs.io/en/master/workflows/ipynb/examples/analysis_plots.html#Holoviews

* There is a good chance you will find this plot better if you are familiar with kernelshark: https://lisa-linux-integrated-system-analysis.readthedocs.io/en/master/trace_analysis.html#lisa.analysis.tasks.TasksAnalysis.plot_task_activation

douglas-raillard-arm, 2021-09-27:
Looking at the code of `plot_task_residency` it seems to be using `plat_info["freq-domains"]`. If for some reason CPU4/5/6/7 are not appearing anywhere in that, that would explain the missing part. `plot_tasks_activation` should not suffer from that, but if the freq domains are inaccurate you will probably encounter other issues.

Leo-Yan, 2022-06-22:
Not sure if it was related with the platform info file or not before.  But I can confirm this issue is dismissed on my Hikey960 board and QEMU emulation platform.  So close it.

closed 2022-06-22 by Leo-Yan