ARM-software/lisa #1826: On Android S, the tracing folder is "/sys/kernel/tracing/"  ,not "/sys/kernel/debug/tracing/"

accsky, 2022-04-13:
Capture ftrace on Android S, how to adapt?

douglas-raillard-arm, 2022-04-13:
Hi @accsky ,

For now the value is unfortunately hardcoded in a couple of places in lisa and devlib. Until that is fixed, maybe you can re-mount tracefs at the expected location if you have root access ? (which I suspect would be needed anyway for tracing)

douglas-raillard-arm, 2022-04-13:
This PRs should solve the issue:
https://github.com/ARM-software/lisa/pull/1828
https://github.com/ARM-software/devlib/pull/590

note: you can use this branch to test both PR at once:
remote: https://github.com/douglas-raillard-arm/lisa.git
branch: detect_tracefs

douglas-raillard-arm, 2022-04-25:
The devlib PR has been merged, LISA install from the git repo with vendored devlib will get the fix as soon as I merge this PR:
https://github.com/ARM-software/lisa/pull/1831

closed 2022-04-25 by douglas-raillard-arm