ARM-software/lisa #1832: when I run typical_experiment.ipynb, The program is stuck at this position::->  2022-05-15 21:35:28,547 INFO    : lisa.target.Target : Freezing all tasks except: sh,adbd,usb,transport,thermal-engine,watchdogd

accsky, 2022-05-15:
**Describe the bug**
A clear and concise description of what the bug is.
[image]

**To Reproduce**
Steps to reproduce the behavior:
* what part of LISA is being used (synthetic tests, WA-related features, etc.)
* what kind of target is being used (linux, android, etc.)
* what code is being run if that is a custom script using LISA APIs

**Expected behavior/what was the purpose of the experiment**
A clear and concise description of what you expected to happen.
Giving the overall purpose of the script you are trying to get to work can help us identify which part of the API you may benefit from much more quickly.

**Logs of the error**
info and debug can be found under the artifacts as INFO.log and DEBUG.log.
The header of the log is useful as it gives plenty of information on python version and similar things.
*Review the log and strip out any credential that you are not comfortable sharing before posting*

**Additional context**
Add any other context about the problem here.

accsky, 2022-05-15:
run lisa-test: stuck here too.
[2022-05-15 22:37:25,362][lisa.target.Target] INFO  Freezing all tasks except: sh,adbd,usb,transport,thermal-engine,watchdogd
[image]

douglas-raillard-arm, 2022-05-16:
Hi @accsky thanks for reporting this issue.

Could you re-run your code with this prepended in the cell and provide the full cell output please ?

```
import logging

from lisa.utils import setup_logging
setup_logging(level=logging.DEBUG)
```

This should show what is going on in more details.

As a temporary workaround, you could disable the cgroups support for the target if that is acceptable for you, using either:
* `Target(..., devlib_excluded_modules=['cgroups'])` if you create the object "manually"
* Setting this key in target_conf.yml:
```yaml
target-conf:
    devlib:
        excluded-modules:
          - cgroups
```

This should prevent the offending code to run, at the cost of not freezing other processes while running the calibration. I strongly recommend disabling any service that would wake up while you run the workload (or the calibration) as preempting rt-app while it calibrates will result in somewhat wrong calibration values.

douglas-raillard-arm, 2022-05-16:
What userspace are you using btw (linux distro, buildroot FS etc) ? That might help me reproduce the issue.

accsky, 2022-05-16:
Ubuntu 22.04 LTS；
You can find the log of the corresponding time point.
[lisa_debug.log](https://github.com/ARM-software/lisa/files/8699241/lisa_debug.log)

douglas-raillard-arm, 2022-05-16:
Could you try with this branch and let me know if the issue is solved ?
repo: https://github.com/douglas-raillard-arm/lisa.git
branch: _test_cgroups_fix

douglas-raillard-arm, 2022-05-16:
Great, thanks for testing. I've opened a PR on devlib to fix the issue:
https://github.com/ARM-software/devlib/pull/592

Once this is merged, I'll update the devlib code we bundle in the LISA repo and close this issue. In the meantime, feel free to use  `_test_cgroups_fix` branch on my fork.

douglas-raillard-arm, 2022-05-17:
Devlib PR has been merged, and the git subtree in LISA has been updated so the issue should be solved
https://github.com/ARM-software/lisa/pull/1834

Please re-open the issue if it turns out there was another problem as well.

closed 2022-05-17 by douglas-raillard-arm