ARM-software/lisa #1838: Can lisa analyze systrace.html or perfetto?

accsky, 2022-05-20:
IN https://lisa-linux-integrated-system-analysis.readthedocs.io/en/master/trace_analysis.html#trace :
[image]
I would like to ask is it effective to analyze systrace with lisa? Because we generally capture systrace or Perfetto offline.

douglas-raillard-arm, 2022-05-20:
Hi @accsky, it is possible but be warned that the trace is actually just a dump of `/sys/kernel/debug/tracing/trace`. AFAIR it suffers from the following flaws:

* timestamps are not accurate, as they are displayed with microsecond accuracy instead of nanoseconds (like trace-cmd report without -t)
* More importantly, it displays the human-readable format of events. While it's great to look at the cute sched_switch pretty printing as a human (and other events), it's a major PITA to parse. As a result, LISA uses the raw printing of events in almost all cases, which forces a "key=value" format.

If only it could simply dump a bit of JSON the world would be a better place.

We still somewhat support it in LISA because:
* we still parse trace-cmd output, and for quite a few events the raw and pretty format are the same, so support was easy to keep.
* It can still be useful to be able to do things with systrace when there is no choice on the format.

But some events will probably not be parsed and we won't go out of our way to try and fix it, especially since `lisa.trace.FtraceCollector` works with android targets too.

douglas-raillard-arm, 2024-02-09:
Closing as question answered

closed 2024-02-09 by douglas-raillard-arm