ARM-software/lisa #1842: running `df_all_events` without executing `trace-cmd` separately for each event

Avidanborisov, 2022-05-27:
I'm trying to use lisa to conveniently parse all the events from a trace.dat file into a DataFrame, with additional columns from the event arguments (such as `func` from `funcgraph_entry`/`funcgraph_exit`) to appear on the DataFrame. Here's an example - 
```
import lisa.trace.Trace
t = lisa.trace.Trace("trace.dat")
df = t.df_all_events(fields_as_cols=["__file", "__cpu", "__comm", "__pid", "func"])
```

Running this on a trace.dat file with ~2 million events and ~14 different kinds of events takes a good few minutes to complete. I've noticed that a separate `trace-cmd` process is getting executed for each event separately (with an `-r event_name` argument), and an additional initial invocation with no events at all.
Is there a way to parse all events together with a single `trace-cmd` invocation, thus reducing time for this to complete?

Thanks!

douglas-raillard-arm, 2022-05-30:
Hi @Avidanborisov,

There are a few things at play here:
* The event parsers (`EventParserBase`) are implemented to only parse one event at a time (even though the API does have a `parse_events()`, but it's a trivial implementation). 
* The `Trace` class is able to spawn multiple parsers in parallel using a `multiprocessing.Pool`, but:
    * `Trace.df_event()` only allows requesting one event, and `Trace.df_events()` is a deprecated alias (although by now it could probably be re-purposed, it's been years)
    * Parsing in parallel is "dangerous" as the peak memory consumption could be quite high.

So that leaves `Trace` with only one path that is currently achieving parallel parsing: When building the `Trace()` object with an explicit list of events, and `max_mem_size=None`.

In order to move forward, you can pass an explicit list of events to `Trace(events=...)`. That will parse in parallel and fix the issue in Note 1. You might want to increase `Trace(max_swap_size=...)` to ensure Note 2 applies.

---

Note 1: Calling `trace.ana.notebook.df_all_events()` without an explicit list of events results in an arbitrary set of events being included (see https://lisa-linux-integrated-system-analysis.readthedocs.io/en/master/trace_analysis.html?highlight=df_all_events#lisa.analysis.notebook.NotebookAnalysis.df_all_events). This is because trace formats don't all have a way of querying the list of events they contain (trace.dat does, but a simple text dump from tracefs does not). This means that `trace.available_event` can only know about events it got queried about with `foobar in trace.available_event`.

Note 2: While not a real solution to your problem, `Trace()` will cache the parsed event dataframes in the `parquet` format. If it does not, you might want to increase `Trace(..., max_swap_size=...)` (defaults to the size of the trace itself). That means subsequent run on the same trace.dat with the same version of LISA will be much faster. Ofc it does not help if you only process your traces once, but interactive notebook exploration is greatly improved by the cache.

Note 3: Once all events are parsed, creating the the `df_all_events()` dataframe itself is somewhat costly (Python string manipulation, heavy load on the GC etc). It's an embarrassingly parallel problem, but would need multiprocessing, meaning pickling the event dataframes, which has a cost in itself. Might need to investigate the out of band data support in pickle v5.

Note 4:
> and an additional initial invocation with no events at all
This one is harmless, it's just the query of CPUs count. We consume the first line of the output then close stdout, leading to `trace-cmd report` immediately exiting without actually parsing the trace.

douglas-raillard-arm, 2022-05-30:
Forgot a note 5: I'm currently working on a "next-gen" analysis setup where:
* trace analysis functions in LISA will be written in Rust
* Ultimately, it could probably link to libtraceevent and consume the trace.dat directly, so no more expensive text regex parsing for the common case
* The analysis functions will be available in LISA's kernel module as well, so they can be ran in-kernel without the need to store a giant trace. This will add flexibility to the offline-trace-only flow we currently have. Debugging of an analysis will therefore be done offline on a small trace in a notebook, which can then be scaled to run in-kernel for hours or even days, where getting an offline trace would be impractical.
* Given that LISA compiles the kernel module on its own and can setup an alpine chroot, it should be perfectly feasible to allow users to provide Rust analysis as well, even e.g. in a dynamic notebook even if users don't have a rust toolchain installed. The dataframe API will remain in place for:
* existing users
* as an interface to plotting libraries
* As a "debugging API" for users to view a dump of intermediate streams used in the Rust analysis such as `trace.ana.tasks.df_task_states()`

closed 2022-10-14 by douglas-raillard-arm