ARM-software/lisa #1846: lisa-test reports failure "EXCEPTION (AttributeError): 'DataFrame' object has no attribute 'event_name'"

Leo-Yan, 2022-06-14:
**Describe the bug**
When I execute the command: lisa-test '*test_task_placement', it reports the failure:
EXCEPTION (AttributeError): 'DataFrame' object has no attribute 'event_name'

**To Reproduce**
Steps to reproduce the behavior:
* lisa-test
* Hikey960 with mainline kernel 5.19
* No any other custom change

**Expected behavior/what was the purpose of the experiment**
n/a

**Logs of the error**

```
[2022-06-14 10:28:26,863][lisa.target.Target] WARNING  Could not freeze userspace: "cgroups" devlib module is necessary
[2022-06-14 10:28:26,889][lisa.wlgen.rta.RTA] INFO  Created workload's run target directory: /root/devlib-target/lisa/wlgen/20220614_102826_592e280c4e184a60abb5c10cd53cfb1e
[2022-06-14 10:28:27,120][lisa.wlgen.rta.RTA] WARNING  CPU capacities will not be updated on this platform
[2022-06-14 10:28:27,133][lisa.wlgen.rta.RTA] INFO  CPU capacities according to rt-app workload: {0: 462, 1: 462, 2: 462, 3: 462, 4: 1024, 5: 1024, 6: 1024, 7: 1024}
[2022-06-14 10:28:34,451][lisa.wlgen.rta.RTA] INFO  Execution start: rt-app /root/devlib-target/lisa/wlgen/20220614_102826_592e280c4e184a60abb5c10cd53cfb1e/rta_energymodelwakemigration.json 2>&1
[2022-06-14 10:29:02,493][lisa.wlgen.rta.RTA] INFO  Wiping target run directory: /root/devlib-target/lisa/wlgen/20220614_102826_592e280c4e184a60abb5c10cd53cfb1e
[2022-06-14 10:29:03,369][lisa.target.Target] INFO  Re-enabling idle states for all domains
[2022-06-14 10:29:05,342][EXEKALL] INFO  Computed EnergyModelWakeMigration.from_target[board=hikey960] UUID=d3056671fb654a62a01055498aee672f
[2022-06-14 10:29:06,931][lisa.platforms.platinfo.PlatformInfo] INFO  Attempting to read kallsyms from target
[2022-06-14 10:29:21,192][EXEKALL] INFO  Computed EnergyModelWakeMigration.from_target[board=hikey960]:EASBehaviour.test_task_placement UUID=3f5859f6813049e8900548ee470df711
[2022-06-14 10:29:21,194][EXEKALL] ERROR  AttributeError: 'DataFrame' object has no attribute 'event_name'
ID: lisa.target.TargetConf:lisa.target.Target.from_conf[board=hikey960](res_dir=lisa.exekall_customize.ExekallArtifactPath.from_expr_data,plat_info=lisa.platforms.platinfo.PlatformInfo):lisa.tests.base.EnergyModelWakeMigration.from_target[board=hikey960](res_dir=lisa.exekall_customize.ExekallArtifactPath.from_expr_data,ftrace_conf=lisa.trace.FtraceConf):lisa.tests.scheduler.eas_behaviour.EASBehaviour.test_task_placement
Traceback (most recent call last):
  File "/home/leoy/Work2/tools/lisa/tools/exekall/exekall/engine.py", line 2857, in genf
    val = self.callable_(**kwargs)
  File "/home/leoy/Work2/tools/lisa/tools/exekall/exekall/engine.py", line 2338, in __call__
    return __UnboundMethod_self__.__wrapped__(*args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/base.py", line 492, in wrapper
    res = func(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 2319, in wrapper
    return f(**kwargs, **dispatched)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/base.py", line 614, in filter_wrapper
    res = wrapped_test(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 541, in wrapper
    raise e
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 527, in catch
    x = f(*args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 492, in test_task_placement
    exp_power = self._get_expected_power_df(nrg_model, capacity_margin_pct)
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 390, in _get_expected_power_df
    task_utils_df = self._get_expected_task_utils_df()
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 263, in _get_expected_task_utils_df
    cols = dict(
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 264, in <genexpr>
    task_util(task, wlgen_task)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 236, in task_util
    df = self.trace.ana.rta.df_phases(task, wlgen_profile=rtapp_profile)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 1679, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 541, in wrapper
    raise e
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4867, in wrapper
    trace = self.trace
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 1679, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 541, in wrapper
    raise e
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4867, in wrapper
    trace = self.trace
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 1679, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 541, in wrapper
    raise e
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4867, in wrapper
    trace = self.trace
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 1679, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 541, in wrapper
    raise e
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4867, in wrapper
    trace = self.trace
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 1679, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 541, in wrapper
    raise e
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4867, in wrapper
    trace = self.trace
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 1679, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 541, in wrapper
    raise e
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 527, in catch
    x = f(*args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/base.py", line 1757, in trace
    return trace.get_view(self.trace_window(trace), clear_base_cache=True)
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/base.py", line 1666, in trace_window
    phase_start_df = trace.ana.rta.df_rtapp_phases_start(
  File "/home/leoy/Work2/tools/lisa/lisa/analysis/_proxy.py", line 59, in wrapper
    return x(**kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/analysis/rta.py", line 446, in df_rtapp_phases_start
    return self._get_rtapp_phases('start', task, wlgen_profile=wlgen_profile)
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/analysis/rta.py", line 295, in _get_rtapp_phases
    df = self.df_rtapp_loop(task, wlgen_profile=wlgen_profile)
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/analysis/rta.py", line 288, in df_rtapp_loop
    df = df.groupby(['__pid', '__comm'], observed=True).apply(f)
  File "/home/leoy/Work2/tools/lisa/.lisa-venv-3.8/lib/python3.8/site-packages/pandas/core/groupby/groupby.py", line 1414, in apply
    result = self._python_apply_general(f, self._selected_obj)
  File "/home/leoy/Work2/tools/lisa/.lisa-venv-3.8/lib/python3.8/site-packages/pandas/core/groupby/groupby.py", line 1455, in _python_apply_general
    values, mutated = self.grouper.apply(f, data, self.axis)
  File "/home/leoy/Work2/tools/lisa/.lisa-venv-3.8/lib/python3.8/site-packages/pandas/core/groupby/ops.py", line 761, in apply
    res = f(group)
  File "/home/leoy/Work2/tools/lisa/lisa/analysis/rta.py", line 272, in f
    pid, comm = df.event_name
  File "/home/leoy/Work2/tools/lisa/.lisa-venv-3.8/lib/python3.8/site-packages/pandas/core/generic.py", line 5583, in __getattr__
    return object.__getattribute__(self, name)
AttributeError: 'DataFrame' object has no attribute 'event_name'

Finished UUID=3f5859f6813049e8900548ee470df711 in 15.85s (cumulative: 129.19s) EnergyModelWakeMigration[board=hikey960]:test_task_placement
                                                                                                                        ^^^^^^^^^^^^^^^^^^^
EXCEPTION (AttributeError): 'DataFrame' object has no attribute 'event_name'
```

douglas-raillard-arm, 2022-06-14:
Hi @Leo-Yan , thanks for reporting that. 

Would you be able to share the artifact folder of exekall as a compressed tar that goes with this test execution so I can reproduce locally ? It is reported when the test starts executing.

Note that these artifacts will contain the target's  credentials.

Leo-Yan, 2022-06-16:
Sure, @douglas-raillard-arm.  Will test it in my tomorrow morning and share back the result.  Thanks for looking into the issue.

douglas-raillard-arm, 2022-06-21:
@Leo-Yan Have you managed to reproduce any issue so far ?

Leo-Yan, 2022-06-21:
Hi @douglas-raillard-arm, 

After upgrading to the latest LISA respository with command lisa-install, and run the command: `exekall run lisa --load-db VALUE_DB.pickle.xz --load-type '*TestBundle' --select '*placement*' --pdb`, then I still can see the issue:

```
[LISAShell for_share_Douglas_v2] \> exekall run lisa --load-db VALUE_DB.pickle.xz --load-type '*TestBundle' --select '*placement*' --pdb
The following expressions will be executed:

OneSmallTask:test_task_placement

Used trace events:
  - sched_switch
  - sched_wakeup
  - sched_wakeup_new
  - task_rename
  - userspace@rtapp_loop

Artifacts dir: /home/leoy/Work2/tools/lisa/results/20220622_01:01:46_1f5bc0dd75be4b6f836089fcfbcd57a6

[2022-06-22 01:01:46,690][EXEKALL] INFO  Iteration #1

#############################################
# Executing: OneSmallTask:test_task_placement
# 
# ID: OneSmallTask:lisa.tests.scheduler.eas_behaviour.EASBehaviour.test_task_placement
# Artifacts: /home/leoy/Work2/tools/lisa/results/20220622_01:01:46_1f5bc0dd75be4b6f836089fcfbcd57a6/OneSmallTask:test_task_placement/5df042ac1a754c0d8e593f05682dc3e8
# UUID: 5df042ac1a754c0d8e593f05682dc3e8
#############################################

[2022-06-22 01:01:46,691][EXEKALL] INFO  Reusing already computed OneSmallTask[board=hikey960] UUID=68fa207bfa6e4998847be4d5a204daaa
[2022-06-22 01:01:48,205][EXEKALL] INFO  Computed OneSmallTask[board=hikey960]:EASBehaviour.test_task_placement UUID=cfb908fe031b4a42b09baf5a521020bf
[2022-06-22 01:01:48,206][EXEKALL] ERROR  Traceback (most recent call last):
  File "/home/leoy/Work2/tools/lisa/tools/exekall/exekall/engine.py", line 2857, in genf
    val = self.callable_(**kwargs)
  File "/home/leoy/Work2/tools/lisa/tools/exekall/exekall/engine.py", line 2338, in __call__
    return __UnboundMethod_self__.__wrapped__(*args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/base.py", line 492, in wrapper
    res = func(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 2319, in wrapper
    return f(**kwargs, **dispatched)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/base.py", line 614, in filter_wrapper
    res = wrapped_test(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 541, in wrapper
    raise e
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 527, in catch
    x = f(*args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 492, in test_task_placement
    exp_power = self._get_expected_power_df(nrg_model, capacity_margin_pct)
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 390, in _get_expected_power_df
    task_utils_df = self._get_expected_task_utils_df()
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 263, in _get_expected_task_utils_df
    cols = dict(
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 264, in <genexpr>
    task_util(task, wlgen_task)
  File "/home/leoy/Work2/tools/lisa/lisa/tests/scheduler/eas_behaviour.py", line 236, in task_util
    df = self.trace.ana.rta.df_phases(task, wlgen_profile=rtapp_profile)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 1679, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 541, in wrapper
    raise e
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4867, in wrapper
    trace = self.trace
  File "/home/leoy/Work2/tools/lisa/lisa/utils.py", line 1679, in wrapper
    return f(self, *args, **kwargs)
[... 40 lines cut ...]
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/analysis/rta.py", line 295, in _get_rtapp_phases
    df = self.df_rtapp_loop(task, wlgen_profile=wlgen_profile)
  File "/home/leoy/Work2/tools/lisa/lisa/trace.py", line 4876, in wrapper
    return f(self, *args, **kwargs)
  File "/home/leoy/Work2/tools/lisa/lisa/analysis/rta.py", line 288, in df_rtapp_loop
    df = df.groupby(['__pid', '__comm'], observed=True).apply(f)
  File "/home/leoy/Work2/tools/lisa/.lisa-venv-3.8/lib/python3.8/site-packages/pandas/core/groupby/groupby.py", line 1414, in apply
    result = self._python_apply_general(f, self._selected_obj)
  File "/home/leoy/Work2/tools/lisa/.lisa-venv-3.8/lib/python3.8/site-packages/pandas/core/groupby/groupby.py", line 1455, in _python_apply_general
    values, mutated = self.grouper.apply(f, data, self.axis)
  File "/home/leoy/Work2/tools/lisa/.lisa-venv-3.8/lib/python3.8/site-packages/pandas/core/groupby/ops.py", line 761, in apply
    res = f(group)
  File "/home/leoy/Work2/tools/lisa/lisa/analysis/rta.py", line 272, in f
    pid, comm = df.event_name
  File "/home/leoy/Work2/tools/lisa/.lisa-venv-3.8/lib/python3.8/site-packages/pandas/core/generic.py", line 5575, in __getattr__
    return object.__getattribute__(self, name)
AttributeError: 'DataFrame' object has no attribute 'event_name'

> /home/leoy/Work2/tools/lisa/.lisa-venv-3.8/lib/python3.8/site-packages/pandas/core/generic.py(5575)__getattr__()
-> return object.__getattribute__(self, name)
(Pdb) 

```

douglas-raillard-arm, 2022-06-21:
Hi @Leo-Yan , thanks for reporting

This is fixed by this pull request I just merged
https://github.com/ARM-software/lisa/pull/1853

Leo-Yan, 2022-06-22:
After I pulled the latest patches from https://github.com/ARM-software/lisa/pull/1853, then I can confirm the issue for "Overlapping resources manipulated in concurrent async tasks" has been fixed.

Thanks a lot for quick fixing, @douglas-raillard-arm !  Let's close this ticket.

closed 2022-06-22 by Leo-Yan