ARM-software/lisa #1902: How to use Lisa to obtain cpu usage and generate a chart

wuhao-1998, 2022-10-16:
Can you give me some ideas？ Thank you
[image]

douglas-raillard-arm, 2022-10-17:
Hi @wuhao-1998 ,

Something like that should do the trick:
* Methods that return "time spent in X state" are have usually "residency" in their name 
* We don't have any info on voltage unfortunately, as no trace event provides that AFAIK, let me know if you have more info on that front.

```
import pandas as pd

from lisa.utils import setup_logging
setup_logging()
from lisa.trace import Trace
from lisa.platforms.platinfo import PlatformInfo

plat_info = PlatformInfo.from_yaml_map('./doc/traces/plat_info.yml')
trace = Trace('./doc/traces/trace.dat', plat_info=plat_info, normalize_time=True)

def get_df(cpu):
    # https://lisa-linux-integrated-system-analysis.readthedocs.io/en/master/trace_analysis.html#lisa.analysis.frequency.FrequencyAnalysis.df_cpu_frequency_residency
    df = trace.ana.frequency.df_cpu_frequency_residency(cpu)
    df = df.copy()

    pct = 100 * (df['total_time'] / trace.time_range)

    # Transpose the dataframe so we get one column per freq instead of one row per freq
    df = pct.to_frame().T

    df.index = [cpu]
    return df

dfs = {
    cpu: get_df(cpu)
    for cpu in range(trace.cpus_count)
}

df = pd.concat(dfs.values())
df = df.fillna(0)

print(df)
breakpoint()
```
It seems that rows don't exactly add up to 100% (between 96% and 99%), which is probably due to not having an event right at the beginning of the trace. Using a window on the trace with `trace = trace[2:3]` fixes that, which is consistent with that explanation.

For the display of the dataframe in jupyter lab notebooks, we use this widget in LISA:
https://panel.holoviz.org/reference/widgets/Tabulator.html
Not sure if it will be able to display a bar like in your example though, but maybe you can vary the cell background color or something like that.

wuhao-1998, 2022-10-18:
Thank you, this gave me ideas and inspiration

wuhao-1998, 2022-10-18:
Thank you, this gave me ideas and inspiration.I thought voltage might get in the energy model.

douglas-raillard-arm, 2022-10-18:
The energy model as modeled in LISA only contains a mapping `freq -> power`, so the voltage is never appearing explicitly, and I'm not sure if any trace event reports the voltage either

wuhao-1998, 2022-10-18:
Yes, it's a formula. power = C（dynamic-power-coefficient : devicetree） * V（Voltage）* V（Voltage） * f (frequency）

wuhao-1998, 2022-11-04:
This can be turned off Thank you

closed 2022-11-18 by wuhao-1998