ARM-software/lisa #1906: The running time of a task at each frequency on each CPU

wuhao-1998, 2022-11-04:
How to calculate the running time of a task at each frequency on each CPU to form a dataframe？
This is an excel sheet I made with lisa, but it is missing a usage rate for each frequency, can you help me? Or can you give me some hints?
[image]

douglas-raillard-arm, 2022-11-04:
From your table it seems like you should be able to replace your `coreN` columns by `(coreN, freqX)` and apply the same recipe ? 

To get this data, you will need to include the CPU frequency to your "source" dataframe you are currently working with:
https://lisa-linux-integrated-system-analysis.readthedocs.io/en/master/trace_analysis.html#lisa.analysis.frequency.FrequencyAnalysis.df_cpus_frequency

I haven't done that in a while but you probably need pandas' groupby() operation on your existing dataframe to get a sub dataframe per CPU, then align the corresponding frequency df on the same index and merge back all the groups. This is similar to join(), but join() is not time-based so for a left row where core N appears, it will dump all the right rows where core N also appeared which is useless (we just want the right row with the timestamp immediately preceding the one on the left row).  

 For  general dataframe manipulation, I recommend the tutorials on the official pandas doc, such as this one:
https://pandas.pydata.org/docs/user_guide/groupby.html
They are quite comprehensive and will probably give you ideas on how to achieve a given overall result.

douglas-raillard-arm, 2022-11-04:
Something like that:
```python
import pandas as pd

df = pd.DataFrame(dict(
    tasks=['a', 'a', 'b', 'b', 'a'],
    cpu=[1, 2, 2, 2, 1],
))
df.index.name = 'time'

freqs = pd.DataFrame(dict(
    cpu=[1, 2, 1],
    freq=[10, 20, 100],
))
freqs.index = [0, 3, 4]
freqs.index.name = 'time'
print(freqs)

freqs = dict(list(freqs.groupby('cpu', group_keys=False)))

def add_freq(subdf):
    # "name" attribute contains the group key
    cpu = subdf.name

    # Align the frequencies for that CPU on the timestamp we care about (aka
    # the subdf index)
    _, _freqs = subdf.align(freqs[cpu], join='right', method='ffill')

    subdf['freq'] = _freqs['freq']
    return subdf

# Pandas seems to merge back the result of apply() following the index, but
# maybe it requires sorting the index again in some situations
res = df.groupby('cpu', group_keys=False).apply(add_freq)
print(res)
```

closed 2022-11-18 by wuhao-1998