ARM-software/lisa #1914: How to control the capture time of trace.dat？

wuhao-1998, 2022-11-21:
Hello, How is the duration of the trace.dat  controlled? By file size? Or time?
I don't seem to notice the configuration when I have time. I studied lisa/trace.py 和 lisa/target.py are not found.
And through many experiments, the trace time is 1-2s, and the size is between 30MB and 45MB, which is not fixed. How can I extend trace.dat?

douglas-raillard-arm, 2022-11-21:
Hi @wuhao-1998 , the typical use is using `FtraceCollector` as a context manager so the is trace captured more "by content" as it will start before the workload of interest and stop after that. If you want to make it "time controlled" you can use e.g. `time.sleep()` as the "workload".

The trace is stored in the kernel ring buffer (controlled by `FtraceCollector(buffer_size=...)`) and is not drained on the go to avoid disturbing scheduler behavior, so that means the buffer needs to be big enough for the amount of records you need (depends on the events and the number of CPUs)

wuhao-1998, 2022-11-22:
Sorry, after your prompt and your own research, I understand that Lisa is capturing according to the events of interest. In this case, it can't capture random tasks in a certain time like systrace?

Your   time sleep() as the "workload".     Sorry, I don't quite understand. Can you give me a simple example?

douglas-raillard-arm, 2022-11-22:
Context managers are nothing more than an `__enter__()` and `__exit__()` methods called at the beginning and the end of a `with` statement [1]. `FtraceCollector` starts capturing events on the target in the kernel ring buffer when `__enter__()` is called and stops when `__exit__()` is called. It then drains the ftrace ring buffer into a trace.dat file and pulls it on the host (=computer running lisa's python code).

It's up to you to decide what happens between `__enter__()` and `__exit__()`. What is typically done is to just run an experiment, e.g. `rt-app` (usually via `lisa.wlgen.rta` API), or anything else using e.g. `target.execute()`, but if you need something time-based, you can use `time.sleep()` to simply wait for some time before stopping the recording.

[1] https://docs.python.org/3/reference/compound_stmts.html#with

closed 2022-12-04 by wuhao-1998