ARM-software/lisa #1948: When I run lisa-test. "Command '<Workload RTA>' returned non-zero exit status 2 "Wrong, how can I solve this?

wuhao-1998, 2023-02-28:
When I run lisa-test. "Command '<Workload RTA>' returned non-zero exit status 2 "Wrong, how can I solve this?

EnergyModelWakeMigration:test_dmesg
EnergyModelWakeMigration:test_slack
EnergyModelWakeMigration:test_task_placement
。。。。。。。。。。。。

douglas-raillard-arm, 2023-02-28:
Hi @wuhao-1998 , I'll need the exact command you ran, the full output and the configuration files you used (target_conf.yml by default) to help you.

wuhao-1998, 2023-02-28:
I ran the lisa-test test and reported the following error to me:
“
######################################
# Executing: CapacitySanity:test_dmesg
# 
# ID: lisa.tests.base.CapacitySanity.from_target:lisa.tests.base.DmesgTestBundleBase.test_dmesg
# Artifacts: /home/wu/lisa/results/20230227_23:44:08_1df9c27a409748efbcc72b1221b5eb63/CapacitySanity:test_dmesg/ec4ddab0f0df4834902b59ce1f5c7e6c
# UUID: ec4ddab0f0df4834902b59ce1f5c7e6c
######################################

[2023-02-27 23:44:49,302][EXEKALL] INFO  Reusing already computed CapacitySanity.from_target UUID=c5617cc6e5d5468f888be5794d2377b1
[2023-02-27 23:44:49,302][EXEKALL] ERROR  subprocess.CalledProcessError: Command '<Workload Sysbench>' returned non-zero exit status 1.
ID: lisa.target.TargetConf:lisa.target.Target.from_conf[board=myboard](res_dir=lisa.exekall_customize.ExekallArtifactPath.from_expr_data,plat_info=lisa.platforms.platinfo.PlatformInfo):lisa.tests.base.CapacitySanity.from_target(res_dir=lisa.exekall_customize.ExekallArtifactPath.from_expr_data,ftrace_conf=lisa.trace.FtraceConf)
Traceback (most recent call last):
  File "/home/wu/lisa/tools/exekall/exekall/engine.py", line 2884, in genf
    val = self.callable_(**kwargs)
  File "/home/wu/lisa/tools/exekall/exekall/engine.py", line 2338, in __call__
    return __UnboundMethod_self__.__wrapped__(*args, **kwargs)
  File "/home/wu/lisa/lisa/utils.py", line 1865, in wrapper
    return wrapped(*args, **kwargs)
  File "/home/wu/lisa/lisa/tests/base.py", line 827, in from_target
    return from_target(*args, **kwargs)
  File "/home/wu/lisa/lisa/tests/base.py", line 1071, in from_target
    bundle = cls._from_target(target, res_dir=res_dir, **kwargs)
  File "/home/wu/lisa/lisa/utils.py", line 2383, in wrapper
    return f(**kwargs, **dispatched)
  File "/home/wu/lisa/lisa/tests/base.py", line 777, in wrapper
    return _real_from_target.__func__(cls, collector=cm, **from_target_kwargs)
  File "/home/wu/lisa/lisa/tests/scheduler/sanity.py", line 54, in _from_target
    output = sysbench(cpus=[cpu], max_duration_s=1).run()
  File "/home/wu/lisa/lisa/wlgen/workload.py", line 497, in run
    x.wait()
  File "/home/wu/lisa/lisa/wlgen/workload.py", line 220, in __exit__
    raise excep
  File "/home/wu/lisa/lisa/wlgen/workload.py", line 208, in __exit__
    action()
  File "/home/wu/lisa/lisa/wlgen/workload.py", line 196, in <lambda>
    action = lambda: self.gen.throw(
  File "/home/wu/lisa/lisa/wlgen/sysbench.py", line 105, in _run
    out = yield self._basic_run()
subprocess.CalledProcessError: Command '<Workload Sysbench>' returned non-zero exit status 1.

Finished UUID=e2c01d16f5a641a0910cb541d648a28d in  (cumulative: 25.02s) CapacitySanity:test_dmesg
                                                                        ^^^^^^^^^^^^^^
EXCEPTION (CalledProcessError): Command '<Workload Sysbench>' returned non-zero exit status 1.
----------------------------------------
  
"

My Android phone, my target_ Conf.yml is as follows
=================================

target-conf:
    kind: android
    name: myboard
    device: 4add6c4c
    username: root                                     
    password: "root"                                   
    
ftrace-conf:
   
platform-info:
    # !include $LISA_HOME/lisa/platforms/juno_r0.yml
    # conf:
        # rtapp:
            # # Calibration mapping of CPU numbers to calibration value for rtapp
            # calib: {}

wuhao-1998, 2023-02-28:
Those above are just a test. I sorted them out. There are many similar ones, about 16 same mistakes, so I want to consult you

douglas-raillard-arm, 2023-02-28:
Ok so there should be `stdout.log` and `stderr.log` files somewhere in a subfolder of `/home/wu/lisa/results/20230227_23:44:08_1df9c27a409748efbcc72b1221b5eb63/CapacitySanity:test_dmesg`.  They will contain the output of the sysbench binary and hopefully explain why it failed. I've never seen sysbench fail to run so far so I don't know what could be causing it.

Feel free to attach these files here if you find anything interesting in them.

douglas-raillard-arm, 2024-01-16:
Closing as this issue hasn't had any activity in ~1year. Feel free to re-open if it still relevant

closed 2024-01-16 by douglas-raillard-arm