ARM-software/lisa #2174: kmodules building failed

Jiaming97, 2024-01-17:
**Describe the bug**
I met with the bug when I tried to build kmodules. 

**To Reproduce**
The version of the LISA is v3.1.0. 

Here is my `target_conf.yml`.
```
target-conf:
  kind: linux
  name: tc
  host: localhost
  port: 8023
  username: root
  password: ""
  strict-host-check: false

  kernel:
    src: /home/jiaguo01/workzone/TC2/buildroot/output/buildroot/tmp_build/linux
    
    modules:
      make-variables:
        CC: clang
        LLVM: -16
      build-env: alpine

  wait-boot:
    enable: false

  devlib:
    file-xfer: scp
    max-async: 1
```
The error log is as follows.
```
[LISAShell lisa] \> lisa-load-kmod --conf target_conf.yml -- echo hello world
[2024-01-17 16:10:05,597][lisa.target.Target] INFO  Target configuration:
+- devlib:
    |- excluded-modules from target_conf.yml (list): []
    |- file-xfer from target_conf.yml (str): scp
    |- max-async from target_conf.yml (int): 1
    +- platform:
        |- class from target_conf.yml (str): devlib.platform.Platform
|- host from target_conf.yml (str): localhost
+- kernel:
    +- modules:
        |- build-env from target_conf.yml (str): alpine
        |- make-variables from target_conf.yml (dict): {'CC': 'clang', 'LLVM': -16}
    |- src from target_conf.yml (str): /home/jiaguo01/workzone/TC2/buildroot/output/buildroot/tmp_build/linux
|- kind from target_conf.yml (str): linux
|- lazy-platinfo from target_conf.yml (bool): False
|- name from target_conf.yml (str): tc
|- password from target_conf.yml (str): <password>
|- port from target_conf.yml (int): 8023
|- strict-host-check from target_conf.yml (bool): False
|- tools from target_conf.yml (list): []
|- username from target_conf.yml (str): root
+- wait-boot:
    |- enable from target_conf.yml (bool): False
    |- timeout from target_conf.yml (int): 10
[2024-01-17 16:10:05,598][lisa.target.Target] INFO  Creating result directory: /workzone/jiaguo01/LISA/lisa/results/Target-tc-20240117_161005.598085
[2024-01-17 16:10:28,367][lisa.target.Target] INFO  Connected to target tc
[2024-01-17 16:10:29,292][lisa.energy_model.EnergyModel.from_target] INFO  Attempting to load EM using LinuxEnergyModel
[2024-01-17 16:10:37,811][lisa.target.Target] INFO  Loading target devlib module cpuidle
[2024-01-17 16:10:47,174][lisa.target.Target] INFO  Loading target devlib module cpufreq
[2024-01-17 16:10:53,988][lisa.target.Target] INFO  Loading target devlib module sched
[2024-01-17 16:11:00,920][sched] INFO  Scheduler sched_domain procfs entries found
[2024-01-17 16:11:00,920][sched] INFO  Detected kernel compiled with SCHED_DEBUG=y
[2024-01-17 16:11:00,920][sched] INFO  CPU capacity sysfs entries found
[2024-01-17 16:11:11,402][lisa.target.Target] INFO  Effective platform information:
[... 112 lines cut ...]
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 357, in apply
    return self.apply_async(func, args, kwds).get()
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 771, in get
    raise self._value
ValueError: Could not load kernel trees:
**from_installed_headers: ValueError: Building from /lib/modules is not supported with the Alpine build environment as /lib/modules might not be self contained (i.e. symlinks pointing outside)**
from_sysfs_headers: CalledProcessError: Command '['chroot', PosixPath('/tmp/tmpla06phsw/overlaid'), 'sh', '-c', 'source /etc/profile && exec make -j36 -C /tmp/tmpxup6g0gx/overlaid -- ARCH=arm64 CC=clang CROSS_COMPILE=aarch64-linux-gnu- KBUILD_MODPOST_WARN=1 LLVM=1 mrproper']' returned non-zero exit status 2.:
make: Entering directory '/tmp/tmpxup6g0gx/overlaid'
Makefile:2: /workzone/jiaguo01/TC2/buildroot/src/linux/Makefile: No such file or directory
**make: *** No rule to make target '/workzone/jiaguo01/TC2/buildroot/src/linux/Makefile'.  Stop.**
make: Leaving directory '/tmp/tmpxup6g0gx/overlaid'

from_proc_config: CalledProcessError: Command '['chroot', PosixPath('/tmp/tmpz32klin8/overlaid'), 'sh', '-c', 'source /etc/profile && exec make -j36 -C /tmp/tmpf2gx9rhu/overlaid -- ARCH=arm64 CC=clang CROSS_COMPILE=aarch64-linux-gnu- KBUILD_MODPOST_WARN=1 LLVM=1 mrproper']' returned non-zero exit status 2.:
make: Entering directory '/tmp/tmpf2gx9rhu/overlaid'
Makefile:2: /workzone/jiaguo01/TC2/buildroot/src/linux/Makefile: No such file or directory
make: *** No rule to make target '/workzone/jiaguo01/TC2/buildroot/src/linux/Makefile'.  Stop.
make: Leaving directory '/tmp/tmpf2gx9rhu/overlaid'

from_user_tree: CalledProcessError: Command '['chroot', PosixPath('/tmp/tmpadj20lo8/overlaid'), 'sh', '-c', 'source /etc/profile && exec make -j36 -C /tmp/tmpse8x466y/overlaid -- ARCH=arm64 CC=clang CROSS_COMPILE=aarch64-linux-gnu- KBUILD_MODPOST_WARN=1 LLVM=1 mrproper']' returned non-zero exit status 2.:
make: Entering directory '/tmp/tmpse8x466y/overlaid'
Makefile:2: /workzone/jiaguo01/TC2/buildroot/src/linux/Makefile: No such file or directory
make: *** No rule to make target '/workzone/jiaguo01/TC2/buildroot/src/linux/Makefile'.  Stop.
make: Leaving directory '/tmp/tmpse8x466y/overlaid'
```
I have checked the path, the file exists.

douglas-raillard-arm, 2024-01-17:
Hi @Jiaming97 , are you using LISA from PyPI or installed from the git repo ? We had a large number of changes since the last PyPI release, so you should try that first.

Then if you still have issues, we will need the logging at debug level to see the output of the build command (LISA is using the `logging` module from the std lib, if you use `lisa.utils.setup_logging()` you can pass `level=logging.DEBUG` to get that).

douglas-raillard-arm, 2024-01-17:
Also if it was working for you before and has broken on very recent kernels, you might be affected by what this PR is fixing (type change of `util_est` member of some structs):
https://github.com/ARM-software/lisa/pull/2161

Jiaming97, 2024-01-18:
Hi @douglas-raillard-arm ,

Thank you for your help. The issue may not be the symlink. I think the problem may be the configurations in `target_conf.yml` . 

Since it is the kernel module building, I thought the prebuilt kernel should be provided in the `kernel/src/`. So I use the following configuration.
```
  kernel:
    src: /home/jiaguo01/workzone/TC2/buildroot/output/buildroot/tmp_build/linux
```
The path is the output of building the linux kernel, which contains the configuration and header files used in the build.  It failed.
It seems that when I change this to the source code. It can build successfully.

douglas-raillard-arm, 2024-01-18:
Yes, `kernel/src` needs to point at the source tree: https://lisa-linux-integrated-system-analysis.readthedocs.io/en/latest/target.html#lisa.target.TargetConf

As a side note, you can probably drop this from your conf btw:
```
      make-variables:
        CC: clang
        LLVM: -16
```
The current code will detect it needs clang and the version from the live target (/proc/config.gz). In alpine build env it will find the package for clang-16 on its own, if you use host env you will need to have clang-16 available. Otherwise it will fallback to the closest version it finds, with a priority to higher versions.

douglas-raillard-arm, 2024-01-18:
Closing as the issue seems resolved

closed 2024-01-18 by douglas-raillard-arm